0x7375646F/Linuwu-Sense #1: Commands doesn't work

yusifvatan, 2025-01-22:
I can download without error. But when I run the commands, it doesn't work. For example:
echo 50.70 | sudo tee /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/fan_speed
tee: '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/fan_speed': Permission denied
50.70

When I try change backlight timeout:
cat /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/backlight_timeout
[bat error]: '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/backlight_timeout': No such file or directory (os error 2)

Not all necessary files are available. There are only these files:
ls /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/ 
./   driver@          four_zoned_kb/  modalias  predator_sense/  subsystem@
../  driver_override  hwmon/          power/    rfkill/          uevent

My device is phn16-71 (2023 version) and kernel version is 6.12.10-arch1-1.

yusifvatan, 2025-01-22:
I don't know much about coding. But I want to help for tests.

0x7375646F, 2025-01-23:
sorry the directories were incorrect please the docummentation again i have updated to the correct onces.
also the payload is seperated by ,

yusifvatan, 2025-01-23:
It really works without any problems now. Thank you very much! But can I add this as a systemd service? for run automatically on restart.

0x7375646F, 2025-01-23:
ya you can i suppose try researching it.
i have yet to implement that feature myself am tight on time rn

yusifvatan, 2025-01-23:
I'm currently trying to make a simple GUI. I'll finish it as soon as possible!

0x7375646F, 2025-01-24:
ok good luck the only gui that is needed rn is for keyboard since its a hassle to configure manually.

yusifvatan, 2025-01-24:
I added most of functions to GUI :  https://github.com/yusifvatan/GUI-for-Linuwu-Sense

0x7375646F, 2025-01-26:
oh damn that's good it seems usable.

0x7375646F, 2025-01-26:
well then i shall close this issue since the issue is resolved.

closed 2025-01-26 by 0x7375646F

sbstratos79, 2025-02-06:
Hi. I am facing a similar issue. I am using PH16-71 (2023 version).
I don't see any four_zone_kb directory.
There's a backlight_timeout file in the predator_sense directory, but setting it's value to 1 didn't seem to have any effect.

[image]

0x7375646F, 2025-02-06:
@sbstratos79  firstly you are not seeing four_zone_kb because your laptop is not whitelisted to use that feature i have specifically said that only phn16-71 is fully supported your model is PH16-71 which has a per key RGB which is currently not supported by this module.
Also the fact that your model is ph16-71 which has per key rgb keyboard could be the reason why your backlight timeout feature is not working. Sorry for the inconvenience i hope other features work as intended.

sbstratos79, 2025-02-06:
> @sbstratos79  firstly you are not seeing four_zone_kb because your laptop is not whitelisted to use that feature i have specifically said that only phn16-71 is fully supported
Yes, I am aware of this. That's why I offered to provide any device specific information that might be needed to enable support for this model.
You've included support for other models in your roadmap and I think I can help with that to some degree.

0x7375646F, 2025-02-06:
> Yes, I am aware of this. That's why I offered to provide any device specific information that might be needed to enable support for this model. You've included support for other models in your [...]

Ya but currently am on tight on time got semester exams comming up :') 
I will try to add per key rgb feature too in future. but not rn.

0x7375646F, 2025-02-06:
If you got discord hmu in discord: sudo.dwg
When i get free time we can chat and integrate the feature.

sbstratos79, 2025-02-06:
Sent you a friend request on discord. Let me know whenever you're free.

Order52, 2025-02-22:
> Hi. I am facing a similar issue. I am using PH16-71 (2023 version). I don't see any `four_zone_kb` directory. There's a `backlight_timeout` file in the `predator_sense` directory, but setting it's value to 1 didn't seem to have any effect.
> 
> [image]
> 
> I know that only PHN16-71 is fully supported. I am willing to provide diagnostic information or any other information that might be useful for enabling support for PH16-71.

same problem seems like there is no support for PH16-71 at the moment.