0x7375646F/Linuwu-Sense #2: Not installing for Predator Helios Neo 16 PHN16-71-59XW

rishi-2406, 2025-02-09:
make[1]: Entering directory '/path/to/linux-headers'  
warning: the compiler differs from the one used to build the kernel  
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0  
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0  
  CC [M]  /path/to/source.o  
/path/to/source.c: In function ‘backlight_init’:  
/path/to/source.c:1769:27: error: ‘BACKLIGHT_POWER_ON’ undeclared (first use in this function)  
 1769 |         bd->props.power = BACKLIGHT_POWER_ON;  
      |                           ^~~~~~~~~~~~~~~~~~  
/path/to/source.c:1769:27: note: each undeclared identifier is reported only once for each function it appears in  

/path/to/source.c: In function ‘wmi_input_setup’:  
/path/to/source.c:2600:49: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Werror=incompatible-pointer-types]  
 2600 |                                                 wmi_notify_handler, NULL);  
      |                                                 ^~~~~~~~~~~~~~~~~~  
      |                                                 |  
      |                                                 void (*)(union acpi_object *, void *)  

In file included from ./include/linux/i2c.h:13,  
                 from ./include/uapi/linux/fb.h:6,  
                 from ./include/linux/fb.h:7,  
                 from ./include/linux/backlight.h:13,  
                 from /path/to/source.c:19:  
./include/linux/acpi.h:401:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(unsigned int,  void *)’} but argument is of type ‘void (*)(union acpi_object *, void *)’  
  401 |                                         wmi_notify_handler handler, void *data);  
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~  

/path/to/source.c: At top level:  
/path/to/source.c:3925:19: error: initialization of ‘int (*)(struct platform_device *)’ from incompatible pointer type ‘void (*)(struct platform_device *)’ [-Werror=incompatible-pointer-types]  
 3925 |         .remove = platform_remove,  
      |                   ^~~~~~~~~~~~~~~  
/path/to/source.c:3925:19: note: (near initialization for ‘platform_driver.remove’)  

cc1: some warnings being treated as errors  
make[3]: *** [scripts/Makefile.build:243: /path/to/source.o] Error 1  
make[2]: *** [/path/to/linux-headers/Makefile:1925: /path/to/source] Error 2  
make[1]: *** [Makefile:240: __sub-make] Error 2  
make[1]: Leaving directory '/path/to/linux-headers'  
make: *** [Makefile:11: all] Error 2

0x7375646F, 2025-02-12:
did ya install your kernel headers?

rishi-2406, 2025-02-15:
yes

> did ya install your kernel headers?

0x7375646F, 2025-02-15:
and what kernel version do you have?

rishi-2406, 2025-02-15:
> and what kernel version do you have?

6.11.0-17-generic

0x7375646F, 2025-02-17:
this kernel module is only tested in 6.12, 6.13 i g 6.11 is not supported prolly.

Order52, 2025-02-21:
Try

`cd Linuwu-Sense`

`make install MDIR=/run/modules/$(uname -r)/kernel/drivers/platform/x86`

closed 2025-04-06 by 0x7375646F