0x7375646F/Linuwu-Sense #3: Acer Nitro v15-41

PaulDvor, 2025-02-17:
Hi, will it work with Acer Nitro v15-41? I only need fan speed. Ideally to get that Quiet, Balanced, performance mode, but I don't even hope it will work.

Can I talk to you somewhere? I coded a while on C++ and right now I'm getting into RUST, I'd like to ask a few questions.

PaulDvor, 2025-02-17:
Thanks, so currently I can set fans to auto or manually set fans speed, right?

0x7375646F, 2025-02-17:
ya but your device needs to be whitelisted

0x7375646F, 2025-02-17:
ok its been added you can check.

PaulDvor, 2025-02-17:
Weird, I compiled the module even tho I had to specify I want to use Clang instead of GCC, it compiled, loaded but says there is no such file or directory when I use any of these paths:

/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense

0x7375646F, 2025-02-17:
can you confirm that the module is loaded? 
lsmod  | grep linuwu_sense
and also 
sudo dmesg

PaulDvor, 2025-02-17:
❯ sudo dmesg | grep lenuwu_sense

~
❯ lsmod | grep linuwu_sense
linuwu_sense           81920  0
platform_profile       12288  2 linuwu_sense,amd_pmf
sparse_keymap          12288  1 linuwu_sense
rfkill                 40960  8 linuwu_sense,bluetooth,cfg80211
i8042                  57344  1 linuwu_sense
video                  81920  3 linuwu_sense,amdgpu,nvidia_modeset
wmi                    36864  3 linuwu_sense,video,wmi_bmof

~
❯ 
sudo dmesg output is way too large but at the end says [   29.954657] linuwu_sense: device state: 0x8

PaulDvor, 2025-02-17:
❯ lsmod | grep linuwu_sense
linuwu_sense           81920  0
platform_profile       12288  2 linuwu_sense,amd_pmf
sparse_keymap          12288  1 linuwu_sense
rfkill                 40960  8 linuwu_sense,bluetooth,cfg80211
i8042                  57344  1 linuwu_sense
video                  81920  3 linuwu_sense,amdgpu,nvidia_modeset
wmi                    36864  3 linuwu_sense,video,wmi_bmof

~
❯ sudo dmesg | grep linuwu_sense
[    6.157577] linuwu_sense: Acer Laptop ACPI-WMI Extras
[    6.157605] linuwu_sense: Function bitmap for Communication Button: 0x1
[   29.954657] linuwu_sense: device state: 0x8

~
❯

PaulDvor, 2025-02-17:
sorry i was dump now it outputs whats needed ls /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense
.rw-r--r-- 4,1k root 17 лют 15:46  battery_calibration
.rw-r--r-- 4,1k root 17 лют 15:46  battery_limiter
.rw-r--r-- 4,1k root 17 лют 15:46  fan_speed
.rw-r--r-- 4,1k root 17 лют 15:46  usb_charging

~
❯

0x7375646F, 2025-02-17:
well lmk if it works

PaulDvor, 2025-02-19:
just checked - all works as it was designed to you're my savior cuz i couldn't play on Linux due to the fact it was overheating. the only thing it doesnt show an actual rpm pf fans only percentage
could you please implement those quiet balanced performance functions
you can't even imagine what ive already tried to be able to change fans speed on linux

PaulDvor, 2025-02-20:
I checked the code and since you have whitelisted my laptop in quirks table shouldn't it have been having platform_peofile in  /sys/firmware/acpi/platform_profile? It only has pm_profile which isn't what I'm looking for. That platform_profile is exactly what I'm looking for being able to switch between those profiles like quiet, balanced, performance and super important turbo mode. 
ChatGPT says I might need to expose interaction with this profile file via sysfs, adding methods into a platform attributes group... ?

0x7375646F, 2025-02-20:
um i'll implement those feature once i get time. Also i DON'T thinkt here is turbo mode in anv series of laptop.

PaulDvor, 2025-02-22:
> um i'll implement those feature once i get time. Also i DON'T think there is turbo mode in anv series of laptop.

Thanks, mate. No problem if turbo isn't an option, but to be honest I don't know what I did but when I was on windows this turbo button had showed up a few times. Many users report it can be activated, but frankly, it's not a priority for me. I'd really like to get these 3 standart modes which - would solve most of the heating issues when playing on Linux. 
Once you get these done you could create a post or smth on Reddit to let users know there finally is an option to control fans on our laptops. Super great bunch of dudes were looking for it...

PaulDvor, 2025-03-05:
Hi, mate. Are there any news on adding those modes? Also it is said that on predator you can switch these modes with a button press - I'd like to ask which button is this?

PaulDvor, 2025-04-06:
Abandoned?

0x7375646F, 2025-04-06:
oh sorry am busy with my life nowadays.

0x7375646F, 2025-04-19:
Hey try the thermal profiles feature i did add support to it but dk if it works yet. @PaulDvor

0x7375646F, 2025-04-19:
You can change the thermal profile from /sys/firmware/acpi/platform_profile
and your supported profiles are listed in /sys/firmware/acpi/platform_profile_choices
low-power quiet balanced balanced-performance performance
eco           quiet balanced performance turbo

PaulDvor, 2025-04-20:
> You can change the thermal profile from /sys/firmware/acpi/platform_profile
> and your supported profiles are listed in /sys/firmware/acpi/platform_profile_choices
> low-power quiet balanced balanced-performance performance
> eco           quiet balanced performance turbo

Yo, mate, thanks a lot, I'll try that shortly and let you know if it works. I really appreciate what u doin' hope u r fine

PaulDvor, 2025-04-22:
> You can change the thermal profile from /sys/firmware/acpi/platform_profile and your supported profiles are listed in /sys/firmware/acpi/platform_profile_choices low-power quiet balanced balanced-performance performance eco quiet balanced performance turbo

Hey, brotha, i think it works which I'm super grateful for. 

these are wwhich availible cat /sys/firmware/acpi/platform_profile_choices
low-power quiet balanced balanced-performance

 echo balanced-performance | sudo tee /sys/firmware/acpi/platform_profile
[sudo] password for paul: 
balanced-performance

when i set this one - fans stopped spinning. I didn't test it in Games yet, but, mate, you are a life saver if it works

PaulDvor, 2025-04-22:
> Hey try the thermal profiles feature i did add support to it but dk if it works yet. @PaulDvor 

It wooooooorks, finally, my laptop won't hot as hell when playing heavy games and it also won't ramp up fans when there is a slight load because I'll switch it to Quiet mode. 
You can't even imagine how happy I am

closed 2025-04-27 by 0x7375646F