0x7375646F/Linuwu-Sense #4: Everything Works except The RGB Keyboard

Order52, 2025-02-21:
First things first, THIS IS GREAT WORK THANK U <3 really needed all these features.

So like I said above Everything works except the keyboard RGB my model is Predator PH16-71

this is what I get trying to set this up is there a way to fix it :(

```
sloppy@bazzite:~$ sloppy@bazzite:~$ echo 4287f5,4287f5,4287f5,4287f5,100 | sudo tee /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb/per_zone_mode
tee: '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb/per_zone_mode': No such file or directory
4287f5,4287f5,4287f5,4287f5,100
sloppy@bazzite:~$ 

bash: sloppy@bazzite:~$: command not found
tee: '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb/per_zone_mode': No such file or directory
bash: tee:: command not found
bash: 4287f5,4287f5,4287f5,4287f5,100: command not found
bash: sloppy@bazzite:~$: command not found
sloppy@bazzite:~$ echo 3,1,100,2,0,0,0 | sudo tee /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb/four_zone_mode
tee: '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb/four_zone_mode': No such file or directory
3,1,100,2,0,0,0
sloppy@bazzite:~$ 
```
```

sloppy@bazzite:~$ /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/predator_sense
bash: /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/predator_sense: Is a directory
sloppy@bazzite:~$ **ls** /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/predator_sense
backlight_timeout    battery_limiter       fan_speed     usb_charging
battery_calibration  boot_animation_sound  lcd_override
sloppy@bazzite:~$ 

```

Order52, 2025-02-21:
More info

sloppy@bazzite:~$ sudo dmesg
[    0.000000] Linux version 6.12.12-207.bazzite.fc41.x86_64 (builder@8dfb3cd029b2) (gcc (GCC) 14.2.1 20250110 (Red Hat 14.2.1-7), GNU ld version 2.43.1-5.fc41) #1 SMP PREEMPT_DYNAMIC Fri Feb 14 06:48:42 UTC 2025
[    0.000000] Command line: BOOT_IMAGE=(hd1,gpt6)/ostree/default-63e61d7274c04fb67ffeb86c4446198790246f1452e5ad30b43e83ba541df887/vmlinuz-6.12.12-207.bazzite.fc41.x86_64 root=UUID=7ec11694-9323-47e3-9ffd-1ab11ae39cf4 rw resume=UUID=b5ea1f5b-7afe-453e-829a-4c7d45f79e99 bluetooth.disable_ertm=1 preempt=full kvm.ignore_msrs=1 kvm.report_ignored_msrs=0 intel_iommu=on iommu=pt rd.driver.pre=vfio-pci vfio_pci.disable_vga=1 kvmfr.static_size_mb=128 vm.overcommit_memory=2 vm.panic_on_oom=2 intel_pstate=active module.sig_enforce=0 ostree=/ostree/boot.1/default/63e61d7274c04fb67ffeb86c4446198790246f1452e5ad30b43e83ba541df887/0
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000040228fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040229000-0x0000000040b28fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040b29000-0x000000004293efff] usable
[    0.000000] BIOS-e820: [mem 0x000000004293f000-0x00000000449defff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000449df000-0x000000004798efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000004798f000-0x0000000047afefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000047aff000-0x0000000047afffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000047b00000-0x000000004bffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000004c400000-0x000000004c5fffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000004d000000-0x00000000527fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000004ad7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] e820: update [mem 0x38e0f018-0x38e33a57] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000038e0f017] usable
[    0.000000] reserve setup_data: [mem 0x0000000038e0f018-0x0000000038e33a57] usable
[    0.000000] reserve setup_data: [mem 0x0000000038e33a58-0x0000000040228fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000040229000-0x0000000040b28fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000040b29000-0x000000004293efff] usable
[    0.000000] reserve setup_data: [mem 0x000000004293f000-0x00000000449defff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000449df000-0x000000004798efff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000004798f000-0x0000000047afefff] ACPI data
[    0.000000] reserve setup_data: [mem 0x0000000047aff000-0x0000000047afffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000047b00000-0x000000004bffffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000004c400000-0x000000004c5fffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000004d000000-0x00000000527fffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x00000004ad7fffff] usable
[    0.000000] efi: EFI v2.8 by INSYDE Corp.
[    0.000000] efi: ACPI=0x47afe000 ACPI 2.0=0x47afe014 TPMFinalLog=0x478a6000 SMBIOS=0x42fb7000 MEMATTR=0x38e45018 ESRT=0x3a2b2018 MOKvar=0x42f63000 RNG=0x47a72018 TPMEventLog=0x47a66018 
[    0.000000] random: crng init done
[    0.000000] efi: Remove mem86: MMIO range=[0xc0000000-0xcfffffff] (256MB) from e820 map
[    0.000000] e820: remove [mem 0xc0000000-0xcfffffff] reserved
[    0.000000] efi: Not removing mem87: MMIO range=[0xfe010000-0xfe010fff] (4KB) from e820 map
[    0.000000] efi: Remove mem89: MMIO range=[0xff000000-0xffffffff] (16MB) from e820 map
[    0.000000] e820: remove [mem 0xff000000-0xffffffff] reserved
[    0.000000] secureboot: Secure boot disabled
[    0.000000] SMBIOS 3.4 present.
[    0.000000] DMI: Acer Predator PH16-71/Discovery_RTX, BIOS V1.16 11/02/2023
[    0.000000] DMI: Memory slots populated: 2/4
[    0.000000] tsc: Detected 2300.000 MHz processor
[    0.000000] tsc: Detected 2304.000 MHz TSC
[    0.000006] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000007] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000012] last_pfn = 0x4ad800 max_arch_pfn = 0x400000000
[... cut ...]
[11327.330085] linuwu_sense: Fan speeds updated: CPU=0, GPU=5
[11332.976760] linuwu_sense: CUSTOM FAN MODE (GPU)
[11332.992944] linuwu_sense: Fan speeds updated: CPU=0, GPU=50
[11338.242226] linuwu_sense: CUSTOM FAN MODE (MIXED)!
[11338.258245] linuwu_sense: Fan speeds updated: CPU=50, GPU=50
[11347.626781] linuwu_sense: MAX FAN MODE!
[11347.633179] linuwu_sense: Fan speeds updated: CPU=100, GPU=100
[11355.932571] linuwu_sense: AUTO FAN MODE!
[11355.938192] linuwu_sense: Fan speeds updated: CPU=0, GPU=0
sloppy@bazzite:~$

Order52, 2025-02-21:
even more the file doesn't seem to exist

sloppy@bazzite:/var/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense$ ls /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/
driver  driver_override  hwmon  modalias  power  predator_sense  rfkill  subsystem  uevent
sloppy@bazzite:/var/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense$ ls /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb
ls: cannot access '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb': No such file or directory
sloppy@bazzite:/var/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense$

0x7375646F, 2025-02-21:
@Order52  your pc seems to be PH16-71 whose rgb keyboard is per key currently, per key rgb is not supported since i don't own a per key rgb laptop. But , I will be trying to implement it once my semester exam ends. The reason you're not seeing four_zoned_kb is  ofc your laptop's keyboard is per key rgb based.

Order52, 2025-02-21:
> [@Order52](https://github.com/Order52) your pc seems to be PH16-71 whose rgb keyboard is per key currently, per key rgb is not supported since i don't own a per key rgb laptop. But , I will be [...]

Oh, okay :( 

Thanks for the response, I hope you'll pass your semester exam.

Thanks for your work, though if you can get the RGB to work on my model it will be great, I'll be happy to provide you with any information you may require. Thanks again.

My email: realalialakber@gmail.com

Order52, 2025-05-27:
Hello,
@0x7375646F 
UPDATE:
I found this project
https://github.com/fuho/rgbkb

and it works on Acer Helios 16 PH16-71 (Changing the RGB on) and it works 100%

can you implement it somehow with Linuwu-Sense><?

TNtejaswi, 2025-06-24:
Hi @Order52, can you please let me know the exact kernel version used to install Linuwu_Sense.