0x7375646F/Linuwu-Sense #5: I'm getting an error trying to build it

Order52, 2025-02-28:
I'm getting this error for some reason now 

```
sloppy@bazzite:~$ uname -r
6.13.5-100.bazzite.fc41.x86_64

sloppy@bazzite:~/.scripts/Acer/PredatorSense$ git clone https://github.com/0x7375646F/Linuwu-Sense.git
cd Linuwu-Sense
make install
Cloning into 'Linuwu-Sense'...
remote: Enumerating objects: 53, done.
remote: Counting objects: 100% (53/53), done.
remote: Compressing objects: 100% (49/49), done.
remote: Total 53 (delta 22), reused 0 (delta 0), pack-reused 0 (from 0)
Receiving objects: 100% (53/53), 52.63 KiB | 460.00 KiB/s, done.
Resolving deltas: 100% (22/22), done.
make -C /lib/modules/6.13.5-100.bazzite.fc41.x86_64/build M=/var/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.13.5-100.bazzite.fc41.x86_64'
make[2]: Entering directory '/var/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense'
  CC [M]  src/linuwu_sense.o
src/linuwu_sense.c: In function ‘acer_thermal_profile_change’:
src/linuwu_sense.c:2122:17: error: too few arguments to function ‘platform_profile_notify’
 2122 |                 platform_profile_notify();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
In file included from src/linuwu_sense.c:22:
/usr/src/kernels/6.13.5-100.bazzite.fc41.x86_64/include/linux/platform_profile.h:46:6: note: declared here
   46 | void platform_profile_notify(struct platform_profile_handler *pprof);
      |      ^~~~~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c: In function ‘acer_platform_probe’:
src/linuwu_sense.c:3827:17: error: too few arguments to function ‘platform_profile_remove’
 3827 |                 platform_profile_remove();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/kernels/6.13.5-100.bazzite.fc41.x86_64/include/linux/platform_profile.h:44:5: note: declared here
   44 | int platform_profile_remove(struct platform_profile_handler *pprof);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c: In function ‘acer_platform_remove’:
src/linuwu_sense.c:3865:17: error: too few arguments to function ‘platform_profile_remove’
 3865 |                 platform_profile_remove();
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/kernels/6.13.5-100.bazzite.fc41.x86_64/include/linux/platform_profile.h:44:5: note: declared here
   44 | int platform_profile_remove(struct platform_profile_handler *pprof);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/kernels/6.13.5-100.bazzite.fc41.x86_64/scripts/Makefile.build:196: src/linuwu_sense.o] Error 1
make[3]: *** [/usr/src/kernels/6.13.5-100.bazzite.fc41.x86_64/Makefile:2024: .] Error 2
make[2]: *** [/usr/src/kernels/6.13.5-100.bazzite.fc41.x86_64/Makefile:263: __sub-make] Error 2
make[2]: Leaving directory '/var/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense'
make[1]: *** [Makefile:263: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.13.5-100.bazzite.fc41.x86_64'
make: *** [Makefile:11: all] Error 2
sloppy@bazzite:~/.scripts/Acer/PredatorSense/Linuwu-Sense$ 

```

0x7375646F, 2025-02-28:
seems like they have made some changes on platform profile in kernel in the version that you are currently on.
the src needs some patches so that it can run.  Above 6.13.5 i think

0x7375646F, 2025-02-28:
well hold on tight i'll create a seperate branch and add the patches rn i have some assignments to complete. 🥲

Order52, 2025-02-28:
> well hold on tight i'll create a seperate branch and add the patches rn i have some assignments to complete. 🥲

Alright 

really thank you for your work

Order52, 2025-03-01:
> well hold on tight i'll create a seperate branch and add the patches rn i have some assignments to complete. 🥲

Fixed it by passing parameters in the in line 2122

platform_profile_notify();

Replaced with

platform_profile_notify(&platform_profile_handler);

And doing the same for line 3827

platform_profile_remove();

To 

platform_profile_remove(&platform_profile_handler);

and 3865, same thing
platform_profile_remove();
To
platform_profile_remove(&platform_profile_handler);

and finaly make

0x7375646F, 2025-03-02:
Pushed to the main branch thanks :D

Order52, 2025-03-02:
> Hm, but its weird iam using 6.13.5-zen1-1zen its working perfectly fine with the older version of the code without the patch. well the main branch will be for those who will install it with the latest kernel i suppose cause the changes has occured in linux kernel.

yeah, idk why after the I updated to the latest Kernel `6.13.5-102.bazzite.fc41.x86_64 ` everything broke in linuwusense and when I tried to rebuild it it was giving me the error above, and the only solution was making the changes that I did above, I have no idea what could've been the cause of this error.

This is what GPT had to say about it 

The issue is due to changes in the Linux kernel's API between your module's original version and kernel 6.13.5-102. Specifically, platform_profile_notify and platform_profile_remove now require a struct platform_profile_handler* argument.

Causes:
**Kernel API Changes:** The functions were updated to require an explicit handler reference.
**Backward Compatibility:** Older function signatures may no longer be supported.
**Stricter Type Checking:** The compiler enforces new function signatures.

**Why It Worked Before:**
Previous kernel versions allowed implicit behavior or had different function definitions.

**Why It Fails Now:**
The new kernel requires an argument your module doesn’t provide, leading to compilation errors.

**Solution:**
Update your module to pass the required argument:

platform_profile_notify(&platform_profile_handler);
platform_profile_remove(&platform_profile_handler);

Preventing Future Issues:
Monitor Kernel Changelogs for API changes.

**Use Kernel Version Checks:**

#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 13, 0)
    platform_profile_notify(&platform_profile_handler);
#else
    platform_profile_notify();
#endif

Test against new kernels early.
Use backports for multi-version support.
These updates ensure your module compiles and runs on the latest kernel.

closed 2025-04-19 by 0x7375646F