0x7375646F/Linuwu-Sense #6: ERROR: could not insert 'linuwu_sense': Input/output error

patrickalima98, 2025-03-23:
Hello, I'm having problems in my PH16-71. I'm using PopOS 22.04 with Kernel 6.12;

My error is:

```
make -C /lib/modules/6.12.10-76061203-generic/build M=/home/patrickalima98/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.10-76061203-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
make[1]: Leaving directory '/usr/src/linux-headers-6.12.10-76061203-generic'
sudo install -d /lib/modules/6.12.10-76061203-generic/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.12.10-76061203-generic/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
modprobe: ERROR: could not insert 'linuwu_sense': Input/output error
make: *** [Makefile:33: install] Error 1

```

Kernel: Linux helios16 6.12.10-76061203-generic #202412060638~1740154617~22.04~b4b3ebc SMP PREEMPT_DYNAMIC Fri F x86_64 x86_64 x86_64 GNU/Linux

And yes I've installed the Kernel headers

BikashWhitehat, 2025-04-06:
same issue in my PHN16-71 machine after some kernel updates
Current Kernel Version - 6.12.20-2-MANJARO (64-bit)

0x7375646F, 2025-04-06:
can you check if this is loaded ?
```
lsmod | grep acer_wmi
```
if it is loaded then you can't just load the linuwu_sense

BikashWhitehat, 2025-04-06:
it lists nothing on execution of above command

0x7375646F, 2025-04-06:
can you show me your dmesg log?
```
sudo dmesg
```
when inserting the kernel module

BikashWhitehat, 2025-04-07:
when i try to install following log is thrown
`linuwu_sense: version magic '6.12.19-1-MANJARO SMP preempt mod_unload ' should be '6.12.20-2-MANJARO SMP preempt mod_unload '`

0x7375646F, 2025-04-07:
Seems like you are compiling the module with old headers ```6.12.19-1``` try upgrading the headers and compile it again with the newer ```6.12.20.-2``` kernel headers

0x7375646F, 2025-04-07:
fixed with new Makefile

BikashWhitehat, 2025-04-07:
thanks a lot