0x7375646F/Linuwu-Sense #8: Getting Error After Kernel 6.14

Order52, 2025-03-27:
Hello,
I'm getting this error after updating to 6.14

sloppy@CachyOS ~/.s/A/P/Linuwu-Sense $ make install
make -C /lib/modules/6.14.0-3-cachyos/build M=/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense modules
make[1]: Entering directory '/usr/lib/modules/6.14.0-3-cachyos/build'
make[2]: Entering directory '/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 19.1.7
  You are using:           gcc (GCC) 14.2.1 20250207
  CC [M]  src/linuwu_sense.o
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-fdebug-info-for-profiling’
gcc: error: unrecognized command-line option ‘-mllvm’
gcc: error: unrecognized command-line option ‘-mllvm’
gcc: error: unrecognized command-line option ‘-improved-fs-discriminator=true’
gcc: error: unrecognized command-line option ‘-fbasic-block-sections=labels’
make[4]: *** [/usr/lib/modules/6.14.0-3-cachyos/build/scripts/Makefile.build:207: src/linuwu_sense.o] Error 1
make[3]: *** [/usr/lib/modules/6.14.0-3-cachyos/build/Makefile:2002: .] Error 2
make[2]: *** [/usr/lib/modules/6.14.0-3-cachyos/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.14.0-3-cachyos/build'
make: *** [Makefile:11: all] Error 2
sloppy@CachyOS ~/.s/A/P/Linuwu-Sense $ make install CC=clang LD=ld.lld
make -C /lib/modules/6.14.0-3-cachyos/build M=/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense modules
make[1]: Entering directory '/usr/lib/modules/6.14.0-3-cachyos/build'
make[2]: Entering directory '/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense'
  CC [M]  src/linuwu_sense.o
src/linuwu_sense.c:778:40: warning: tentative definition of variable with internal linkage has incomplete non-array type 'struct platform_profile_handler' [-Wtentative-definition-incomplete-type]
  778 | static struct platform_profile_handler platform_profile_handler;
      |                                        ^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/linuwu_sense.c:2001:27: error: incomplete definition of type 'struct platform_profile_handler'
 2001 |                 platform_profile_handler.profile_get =
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/linuwu_sense.c:2003:27: error: incomplete definition of type 'struct platform_profile_handler'
 2003 |                 platform_profile_handler.profile_set =
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/linuwu_sense.c:2007:28: error: incomplete definition of type 'struct platform_profile_handler'
 2007 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/linuwu_sense.c:2009:28: error: incomplete definition of type 'struct platform_profile_handler'
 2009 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/linuwu_sense.c:2011:28: error: incomplete definition of type 'struct platform_profile_handler'
 2011 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/linuwu_sense.c:2013:28: error: incomplete definition of type 'struct platform_profile_handler'
 2013 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/linuwu_sense.c:2015:28: error: incomplete definition of type 'struct platform_profile_handler'
 2015 |                         platform_profile_handler.choices);
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
src/linuwu_sense.c:2017:60: error: too few arguments to function call, expected 4, have 1
 2017 |                 err = platform_profile_register(&platform_profile_handler);
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~                          ^
/usr/lib/modules/6.14.0-3-cachyos/build/include/linux/platform_profile.h:50:16: note: 'platform_profile_register' declared here
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                ^                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   51 |                                          void *drvdata,
      |                                          ~~~~~~~~~~~~~~
   52 |                                          const struct platform_profile_ops *ops);
      |                                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c:2122:27: error: too few arguments to function call, single argument 'dev' was not specified
 2122 |                 platform_profile_notify();
      |                 ~~~~~~~~~~~~~~~~~~~~~~~ ^
/usr/lib/modules/6.14.0-3-cachyos/build/include/linux/platform_profile.h:58:6: note: 'platform_profile_notify' declared here
   58 | void platform_profile_notify(struct device *dev);
      |      ^                       ~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c:3827:27: error: too few arguments to function call, single argument 'dev' was not specified
 3827 |                 platform_profile_remove();
      |                 ~~~~~~~~~~~~~~~~~~~~~~~ ^
/usr/lib/modules/6.14.0-3-cachyos/build/include/linux/platform_profile.h:53:5: note: 'platform_profile_remove' declared here
   53 | int platform_profile_remove(struct device *dev);
      |     ^                       ~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c:3865:27: error: too few arguments to function call, single argument 'dev' was not specified
 3865 |                 platform_profile_remove();
      |                 ~~~~~~~~~~~~~~~~~~~~~~~ ^
/usr/lib/modules/6.14.0-3-cachyos/build/include/linux/platform_profile.h:53:5: note: 'platform_profile_remove' declared here
   53 | int platform_profile_remove(struct device *dev);
      |     ^                       ~~~~~~~~~~~~~~~~~~
src/linuwu_sense.c:778:40: error: tentative definition has type 'struct platform_profile_handler' that is never completed
  778 | static struct platform_profile_handler platform_profile_handler;
      |                                        ^
src/linuwu_sense.c:778:15: note: forward declaration of 'struct platform_profile_handler'
  778 | static struct platform_profile_handler platform_profile_handler;
      |               ^
1 warning and 12 errors generated.
make[4]: *** [/usr/lib/modules/6.14.0-3-cachyos/build/scripts/Makefile.build:207: src/linuwu_sense.o] Error 1
make[3]: *** [/usr/lib/modules/6.14.0-3-cachyos/build/Makefile:2002: .] Error 2
make[2]: *** [/usr/lib/modules/6.14.0-3-cachyos/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/home/sloppy/.scripts/Acer/PredatorSense/Linuwu-Sense'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.14.0-3-cachyos/build'
make: *** [Makefile:11: all] Error 2
sloppy@CachyOS ~/.s/A/P/Linuwu-Sense $

0x7375646F, 2025-04-06:
hm could be the error cause your kernel is compiled in clang but you're compiling the kernel module in gcc.

0x7375646F, 2025-04-07:
hm probably there are some more changes in 6.14 i'll have to look into it.

friday06, 2025-04-15:
> > hm could be the error cause your kernel is compiled in clang but you're compiling the kernel module in gcc.
> 
> how do I fix it ?

i switched to gcc kernel, its still not able to build.

0x7375646F, 2025-04-16:
the code has been patched up for kernel version 6.14 you can try now.

friday06, 2025-04-19:
> the code has been patched up for kernel version 6.14 you can try now.

i think, some think else broke while compiling on my nitro-V with gcc kernel, following error occur Setting permissions via tmpfiles...
Detected model directory: nitro_sense
Failed to create file /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense/backlight_timeout: Permission denied
Failed to create file /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense/boot_animation_sound: Permission denied
Failed to create file /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense/lcd_override: Permission denied
Failed to open path '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb': No such file or directory
Failed to open path '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb': No such file or directory
make: *** [Makefile:53: install] Error 73

0x7375646F, 2025-04-19:
ugh, i think its the problem with the permission management that is apparently added by @kumarvivek1752  for his GUI for linuwu_sense, the kernel module should have compiled and set-uped you can check if the kernel module is loaded by `lsmod | grep linuwu_sense`

friday06, 2025-04-19:
> ugh, i think its the problem with the permission management that is apparently added by [@kumarvivek1752](https://github.com/kumarvivek1752) for his GUI for linuwu_sense, the kernel module should have compiled and set-uped you can check if the kernel module is loaded by `lsmod | grep linuwu_sense`

this part still bothers me, The kernel was built by: clang version 19.1.7
  You are using:           gcc (GCC) 14.2.1 20250207
  CC [M]  src/linuwu_sense.o

0x7375646F, 2025-04-19:
can you give me the output for
```
lsmod | grep linuwu_sense
```

friday06, 2025-04-19:
here you go : 
Linuwu_sense           81920  0
platform_profile       20480  1 linuwu_sense
sparse_keymap          12288  1 linuwu_sense
rfkill                 40960  10 linuwu_sense,iwlmvm,bluetooth,cfg80211
i8042                  57344  1 linuwu_sense
video                  81920  4 linuwu_sense,xe,i915,nvidia_modeset
wmi                    32768  3 linuwu_sense,video,wmi_bmof

0x7375646F, 2025-04-19:
It seems to be working.

closed 2025-04-19 by 0x7375646F