0x7375646F/Linuwu-Sense #9: Turbo button only works once

meowdroider, 2025-04-06:
I am on PHN16-72 and the turbo button works after some modification in the code(changing model name) but it only works once. When i plug in the charger or plug it out then it also works but only once. After that how much i press it, won't work.

meowdroider, 2025-04-06:
I am on arco linux, linux kernel 6.13.8.

meowdroider, 2025-04-06:
And btw, how to get IDs of methods i am trying to get the logo rgb working.

0x7375646F, 2025-04-07:
hm its probably due to the fact that your version is PHN16-72 where turbo button event could be handleed differently.
To get the ids of the method i used wmi_explorer in *windows* https://github.com/vinaypamnani/wmie2/releases/tag/v2.0.0.2

meowdroider, 2025-04-09:
Can we implement the following logic for this model:
Make a file with the name of the next profile.
When button presses:
   make the default profile= profile in the file.
   Reset the driver (bcz after that the button starts working for 1 more time) 
   Then  switch profile of laptop to the one in the file.
   Replace the profile in that file to be the next profile.
Then when we press the button again the cycle will repeat and the profile will change and the driver resets so infinite button presses.

meowdroider, 2025-04-10:
Or i have another logic 
When button presses:
   change the profile
   set the default profile to the profile adopted after press
   reset the driver and since after resetting the default profile will be the current profile so the lights won't flash twice
In this way it will work flawlessly

meowdroider, 2025-04-10:
how to get wmid like the numbers

0x7375646F, 2025-04-16:
hm you mean the payloads for the wmiid?

meowdroider, 2025-04-17:
idk what you call it but it is:
 #define ACER_WMID_SET_GAMING_LED_METHODID **2**
the 2 here

meowdroider, 2025-04-17:
and btw SetGamingLEDBehavior changes the logo rgb style on the back i want to find its method id. Also i want to know what value should i give it to chanhe the style should i run a loop in poweshell that checks all the values?

meowdroider, 2025-04-17:
ok so SetGamingLEDBehavior 131073(0x20001) is static, 196609(0x30001) is breathing and 262145(0x40001) is breathing effect but the main thing is still left that what id represents SetGamingLEDBehavior in linux. so close

meowdroider, 2025-04-17:
I got it working its id was 10:
#define ACER_WMID_SET_GAMING_LED_BEHAVIOR_METHODID 10

meowdroider, 2025-04-19:
🔧 Turbo Button & /sys/firmware/acpi/platform_profile Explained
When I press the Turbo (performance mode) button on my laptop, it visually switches to performance, but the actual system file:
/sys/firmware/acpi/platform_profile
still says "balanced".

Now here's the interesting part:

If I open that file and press Ctrl+S (save) — it switches back to "balanced" mode(or any other mode that i save it as).

After that, I can press the Turbo button again, for one more time.

💡 What This Means
The Turbo button is probably:

Changing something at a firmware level (like ACPI or WMI),

But not updating the  platform profile file and something else which is supposed to tell that the profile has changed .

meowdroider, 2025-04-19:
so basically when it goes to performance mode for example it goes in that mode but system thinks it is still in balanced mode so when i press it again, it tries to switch again to performance mode but it is already there so it seems like nothing is happening.

0x7375646F, 2025-04-19:
uhm I have no idea why that happens to your modle can you add a logging message on
https://github.com/0x7375646F/Linuwu-Sense/blob/main/src/linuwu_sense.c#L2411
and press the button and see what gets logged?

meowdroider, 2025-04-19:
It only gives log for the first time driver loads no log on button press.

0x7375646F, 2025-04-19:
Ya add the logger there compile it and then try pressing the button and see what it logs

meowdroider, 2025-04-19:
static acpi_status acer_predator_state_restore(int value){
	acpi_status status;
+    pr_info("Sir respectfully Restoring predator state for %s power mode\n", value ? "AC" : "battery");
	status = WMI_gaming_execute_u64(ACER_WMID_SET_GAMING_MISC_SETTING_METHODID, 
									value == 0 ? current_states.battery_state.thermal_profile : current_states.ac_state.thermal_profile, NULL);
	if (ACPI_FAILURE(status))
		return AE_ERROR;

	status = acer_set_fan_speed(value == 0 ? current_states.battery_state.cpu_fan_speed : current_states.ac_state.cpu_fan_speed, 
								value == 0 ? current_states.battery_state.gpu_fan_speed : current_states.ac_state.gpu_fan_speed);
	if(ACPI_FAILURE(status)){
		return AE_ERROR;
	} 

	return AE_OK;
}

meowdroider, 2025-04-19:
sudo dmesg | grep respectfully

0x7375646F, 2025-04-19:
?

meowdroider, 2025-04-19:
did i do it right

meowdroider, 2025-04-19:
i don't know C i used ai lol

0x7375646F, 2025-04-20:
Fixed with the new update.

closed 2025-04-20 by 0x7375646F