0x7375646F/Linuwu-Sense #12: it still shows up

friday06, 2025-04-23:
Hey, i saw the configured the code to fix : 
Failed to create file /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense/backlight_timeout: Permission denied
Failed to create file /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense/boot_animation_sound: Permission denied
Failed to create file /sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/nitro_sense/lcd_override: Permission denied
Failed to open path '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb': No such file or directory
Failed to open path '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/four_zoned_kb': No such file or directory

From Showing Up, but it still does, Running this on My Acer Nitro V15

0x7375646F, 2025-04-23:
Well ill have to fix the makefile 🤣 i did but i g its not correctly checking which files to give the permission to.
Since, the permission granting part is done after the compilaiton and installation its causing the problem. The module does work just the permission granting part is messed up.

friday06, 2025-04-24:
okay ig it's a feature then not a bug 💯