0x7375646F/Linuwu-Sense #15: Could not detect predator_sense or nitro_sense in sysfs.

Hoag11, 2025-05-05:
I'm running Fedora 42 on Predator Triton 300 SE PT314-51s
When I install, i have this warning and can't use anything
out put of `uname -r `: 
6.14.0-63.fc42.x86_64
output when install: 
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense clean
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
  CLEAN   Module.symvers
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 15.0.1 20250228 (Red Hat 15.0.1-0)
  You are using:           gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  CC [M]  .module-common.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
Skipping BTF generation for src/linuwu_sense.ko due to unavailability of vmlinux
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 15.0.1 20250228 (Red Hat 15.0.1-0)
  You are using:           gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)
make[2]: Leaving directory '/home/hoang/Downloads/DAMX-0.4.6b/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
sudo install -d /lib/modules/6.14.0-63.fc42.x86_64/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.14.0-63.fc42.x86_64/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
Created symlink '/etc/systemd/system/multi-user.target.wants/linuwu_sense.service' → '/etc/systemd/system/linuwu_sense.service'.
Setting up group and permissions...
Detected user: hoang
sudo usermod -aG linuwu_sense hoang
Setting permissions via tmpfiles...
Warning: Could not detect predator_sense or nitro_sense in sysfs.
Module linuwu_sense installed and configured to load at boot.

Lightdrew, 2025-05-05:
I'm not really great at coding but I wanted to share my solution as I had this exact issue, it turns out that the source file `linuwu_sense.c` lists the various devices it supports in the table `static const struct dmi_system_id acer_quirks[] __initconst`, what worked for me was navigating to it and adding my own device, here's what it would look for you:
```c
static const struct dmi_system_id acer_quirks[] __initconst = {
  {
          .callback = dmi_matched,
          .ident = "Acer Predator Triton 300 SE PT314-51s",
          .matches = {
              DMI_MATCH(DMI_SYS_VENDOR, "Acer"),
              DMI_MATCH(DMI_PRODUCT_NAME, "YOUR_PRODUCT_NAME"), //you can find it using 'dmidecode -s system-product-name'
          },
          .driver_data = &quirk_acer_predator_v4, 
         //there's various quirks you can look at and choose depending on your features, 
         //for my own model i just ended up editing one of the preexisting ones as none were a perfect match
  },
  //other devices
}
  ```
Hope this helps! I hope we'll get the ability to include a custom quirk in the future in case there aren't any matching devices in the table if possible :)

Hoag11, 2025-05-06:
thanks you, i'll try

Hoag11, 2025-05-06:
now it's worked but i can only use battery Limiter, Boot Animation Sound but can't use fan speed and RGB, maybe because my RGB is 3-zone

PXDiv, 2025-05-06:
I checked your model and it seems like you have one of the older acer models with the old predator sense app which instead of WMI methods, use the EC registers to control the fan speeds and performance mode. Then you may have to use the nbfc project https://github.com/nbfc-linux/nbfc-linux if it supports that