0x7375646F/Linuwu-Sense #19: not compatible for kernel 6.14 in fedora 42

rickygpetmalu, 2025-05-14:
for the developer can you make an update where it was compatible for kernel 6.14 of fedora 42 im using acer nitro v15 thanks

rickygpetmalu, 2025-05-19:
rickygmalupit@fedora:~$ git clone https://github.com/0x7375646F/Linuwu-Sense.git 
cd Linuwu-Sense
make install
Cloning into 'Linuwu-Sense'...
remote: Enumerating objects: 110, done.
remote: Counting objects: 100% (69/69), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 110 (delta 40), reused 58 (delta 35), pack-reused 41 (from 1)
Receiving objects: 100% (110/110), 98.75 KiB | 1.08 MiB/s, done.
Resolving deltas: 100% (51/51), done.
make -C /lib/modules/6.14.6-300.fc42.x86_64/build M=/home/rickygmalupit/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.14.6-300.fc42.x86_64'
make[2]: Entering directory '/home/rickygmalupit/Linuwu-Sense'
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  CC [M]  .module-common.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
Skipping BTF generation for src/linuwu_sense.ko due to unavailability of vmlinux
make[2]: Leaving directory '/home/rickygmalupit/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.6-300.fc42.x86_64'
[sudo] password for rickygmalupit: 
sudo install -d /lib/modules/6.14.6-300.fc42.x86_64/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.14.6-300.fc42.x86_64/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
Setting up group and permissions...
Detected user: rickygmalupit
sudo usermod -aG linuwu_sense rickygmalupit
Setting permissions via tmpfiles...
ls: cannot access '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/': No such file or directory
Warning: Could not detect predator_sense or nitro_sense in sysfs.
Module linuwu_sense installed and configured to load at boot.
rickygmalupit@fedora:~/Linuwu-Sense$

Pranav-Talmale, 2025-05-21:
### TL;DR The module works perfectly with kernel 6.14
the module is perfectly compatible with kernel 6.14, I am running the module fine on my machine which uses:
**6.14.7-5-cachyos**

Since my kernel is compiled using clang, I always have to use a different build command to build the kernel module:
```
make CC=clang LD=ld.lld LLVM=1 LLVM_IAS=1 install
```

For Reference, I have an Acer Predator Helios Neo 16 (PHN16-71)

0x7375646F, 2025-05-27:
May I know your full device info not just v15.
Also, does this module work on your device with other distro?

rickygpetmalu, 2025-05-27:
this is the result that when im installing it on fedora 42
rickygmalupit@fedora:~$ git clone https://github.com/0x7375646F/Linuwu-Sense.git 
cd Linuwu-Sense
make install
Cloning into 'Linuwu-Sense'...
remote: Enumerating objects: 133, done.
remote: Counting objects: 100% (92/92), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 133 (delta 52), reused 71 (delta 40), pack-reused 41 (from 1)
Receiving objects: 100% (133/133), 102.57 KiB | 1.05 MiB/s, done.
Resolving deltas: 100% (63/63), done.
make -C /lib/modules/6.14.6-300.fc42.x86_64/build M=/home/rickygmalupit/Linuwu-Sense modules
make[1]: Entering directory '/usr/src/kernels/6.14.6-300.fc42.x86_64'
make[2]: Entering directory '/home/rickygmalupit/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 15.1.1 20250425 (Red Hat 15.1.1-1)
  You are using:           gcc (GCC) 15.1.1 20250521 (Red Hat 15.1.1-2)
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  CC [M]  .module-common.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
Skipping BTF generation for src/linuwu_sense.ko due to unavailability of vmlinux
make[2]: Leaving directory '/home/rickygmalupit/Linuwu-Sense'
make[1]: Leaving directory '/usr/src/kernels/6.14.6-300.fc42.x86_64'
sudo install -d /lib/modules/6.14.6-300.fc42.x86_64/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.14.6-300.fc42.x86_64/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
Created symlink '/etc/systemd/system/multi-user.target.wants/linuwu_sense.service' → '/etc/systemd/system/linuwu_sense.service'.
Setting up group and permissions...
Detected user: rickygmalupit
sudo usermod -aG linuwu_sense rickygmalupit
Setting permissions via tmpfiles...
ls: cannot access '/sys/module/linuwu_sense/drivers/platform:acer-wmi/acer-wmi/': No such file or directory
Warning: Could not detect predator_sense or nitro_sense in sysfs.
Module linuwu_sense installed and configured to load at boot.
rickygmalupit@fedora:~/Linuwu-Sense$