0x7375646F/Linuwu-Sense #49: CachyOS - Fails to make install

war59312, 2025-08-30:
Hey,

Thanks for this.

Sadly fails to make install on CachyOS:

```
git clone https://github.com/0x7375646F/Linuwu-Sense.git
  cd Linuwu-Sense
  make install
Cloning into 'Linuwu-Sense'...
remote: Enumerating objects: 151, done.
remote: Counting objects: 100% (64/64), done.
remote: Compressing objects: 100% (25/25), done.
remote: Total 151 (delta 53), reused 39 (delta 39), pack-reused 87 (from 2)
Receiving objects: 100% (151/151), 139.92 KiB | 3.78 MiB/s, done.
Resolving deltas: 100% (71/71), done.
make -C /lib/modules/6.16.4-3-cachyos/build M=/home/will/Linuwu-Sense modules
make[1]: Entering directory '/usr/lib/modules/6.16.4-3-cachyos/build'
make[2]: Entering directory '/home/will/Linuwu-Sense'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 20.1.8
  You are using:           gcc (GCC) 15.2.1 20250813
  CC [M]  src/linuwu_sense.o
gcc: error: unrecognized command-line option ‘-mretpoline-external-thunk’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
gcc: error: unrecognized command-line option ‘-fdebug-info-for-profiling’
gcc: error: unrecognized command-line option ‘-mllvm’
gcc: error: unrecognized command-line option ‘-mllvm’
gcc: error: unrecognized command-line option ‘-improved-fs-discriminator=true’
gcc: error: unrecognized command-line option ‘-fbasic-block-address-map’
make[4]: *** [/usr/lib/modules/6.16.4-3-cachyos/build/scripts/Makefile.build:287: src/linuwu_sense.o] Error 1
make[3]: *** [/usr/lib/modules/6.16.4-3-cachyos/build/Makefile:2008: .] Error 2
make[2]: *** [/usr/lib/modules/6.16.4-3-cachyos/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/home/will/Linuwu-Sense'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.16.4-3-cachyos/build'
make: *** [Makefile:12: all] Error 2
```

Best,

Will

Lightdrew, 2025-08-30:
@war59312 I had the same issues with the clang-compiled versions of the CachyOS kernels. Fortunately (at least in my case) the fix was enabling the LLVM parameter when installing like this: `make install LLVM=1`. Hope it helps!

war59312, 2025-08-30:
Awesome, that worked, thank you very much @Lightdrew 

```
uname -a
Linux cachyos-nitro 6.16.4-3-cachyos #1 SMP PREEMPT_DYNAMIC Fri, 29 Aug 2025 18:30:55 +0000 x86_64 GNU/Linux
```

```
make install LLVM=1
make -C /lib/modules/6.16.4-3-cachyos/build M=/home/will/Linuwu-Sense modules
make[1]: Entering directory '/usr/lib/modules/6.16.4-3-cachyos/build'
make[2]: Entering directory '/home/will/Linuwu-Sense'
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  CC [M]  .module-common.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
make[2]: Leaving directory '/home/will/Linuwu-Sense'
make[1]: Leaving directory '/usr/lib/modules/6.16.4-3-cachyos/build'
[sudo] password for will: 
sudo install -d /lib/modules/6.16.4-3-cachyos/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.16.4-3-cachyos/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
Created symlink '/etc/systemd/system/multi-user.target.wants/linuwu_sense.service' → '/etc/systemd/system/linuwu_sense.service'.
Setting up group and permissions...
Detected user: will
sudo usermod -aG linuwu_sense will
Setting permissions via tmpfiles...
Warning: Could not detect predator_sense or nitro_sense in sysfs.
Module linuwu_sense installed and configured to load at boot.
```

```
lsmod | grep linuwu_sense
linuwu_sense           94208  0
sparse_keymap          12288  1 linuwu_sense
platform_profile       16384  2 linuwu_sense,amd_pmf
rfkill                 45056  10 linuwu_sense,mt7921e,bluetooth,cfg80211
video                  81920  3 linuwu_sense,amdgpu,nvidia_modeset
wmi                    36864  3 linuwu_sense,video,wmi_bmof
```

war59312, 2025-08-30:
And with patch https://github.com/0x7375646F/Linuwu-Sense/pull/37 installed:

```
make install LLVM=1
make -C /lib/modules/6.16.4-3-cachyos/build M=/home/will/Linuwu-Sense modules
make[1]: Entering directory '/usr/lib/modules/6.16.4-3-cachyos/build'
make[2]: Entering directory '/home/will/Linuwu-Sense'
  CC [M]  src/linuwu_sense.o
  MODPOST Module.symvers
  CC [M]  src/linuwu_sense.mod.o
  LD [M]  src/linuwu_sense.ko
  BTF [M] src/linuwu_sense.ko
make[2]: Leaving directory '/home/will/Linuwu-Sense'
make[1]: Leaving directory '/usr/lib/modules/6.16.4-3-cachyos/build'
sudo install -d /lib/modules/6.16.4-3-cachyos/kernel/drivers/platform/x86
sudo install -m 644 src/linuwu_sense.ko /lib/modules/6.16.4-3-cachyos/kernel/drivers/platform/x86
sudo depmod -a
sudo modprobe linuwu_sense
Created symlink '/etc/systemd/system/multi-user.target.wants/linuwu_sense.service' → '/etc/systemd/system/linuwu_sense.service'.
Setting up group and permissions...
Detected user: will
sudo usermod -aG linuwu_sense will
Setting permissions via tmpfiles...
Detected model directory: nitro_sense
Module linuwu_sense installed and configured to load at boot.
```

Perfect! 🥇

jamesbond3142, 2025-09-09:
> [linuwu_sense.c](https://github.com/user-attachments/files/22059230/linuwu_sense.c)
> 
> Edited file with changes as noted above.

Wow this is perfect, thank you!!!

Lightdrew, 2025-11-26:
@KuroiJasp3r try installing with the LLVM utilities flag enabled if you have not already - `make install LLVM=1`